aurora-opensource/xviz #552: Can support typescript or 'd.ts' file?

hookex, 2019-12-05:


twojtasz, 2019-12-05:
@hookex we don't have typescript defintions currently.  There was internal discussion of flow vs typescript but I simply don't have enough context and the priority for this simply hasn't been high enough.

If you are familiar and could get the ball rolling I would be happy to support you.

hookex, 2019-12-06:
Out team mainly use typescript for programming, and have been following AVS for serveral months. 
Our application is inspired by AVS, thanks for your great job! 
But reading the XVIZ's source code is diffcult without `Interface`、`Strong Type`.

Out visualiiztion team is growing and busy, looking forward to contributing code to AVS in the near future.