aurora-opensource/xviz #554: Why 'vehicle_pose' must be defined?

hookex, 2019-12-05:
Sorry, I don't understant this setup.

My application scene is for warehouse that support a lot of robots(like vehicle)'s movement,does they  confict?

hookex, 2019-12-05:
Is there any example that show several vechicles at the same time?

twojtasz, 2019-12-05:
@hookex great question.  The problem was one of use-case exposure.  Originally we had a hero-vehicle or ego-vehicle and didn't operation with multi-vehicles in mind.

The good news is that we have been making changes to expand and support this use-cases in a much more general fashion.  This includes no vehicle and multiple vehicles.

Not everything is in place to make this as easy as it should be.  If you can provide more info for your use-case I can offer up some options in the meantime.

Specifically, the structure of your data relative to the vehicles.  in streetscape.gl the camera by default tracks the pose defined by `/vehicle_pose`.  Also, if the coordinate system is VEHICLE_RELATIVE, it is relative to the `/vehicle_pose`.

Is your data described in a shared/global references frame, or relative to each individual vehicle?
Do you need the camera to track vehicles?

hookex, 2019-12-06:
Here's a recording of our app.
[warehouse.zip](https://github.com/uber/xviz/files/3930659/warehouse.zip)

Now, our use-case use global camera to observe  overall warehouse.

Previously, we developed out application inspired by deck.gl and XVIZ, but not use these two module directly. And now, we plan use XVIZ module as our data layer. This is because data protocal and service is so import and difficlut. If XVIZ don't  support our use-case,  we will clone and modify it.

But i seem to have found way(data) to meet our use-case.
For example:

Per pose streamID path contains robotID.  I know this does not conform to the XVIZ spcification, but it is solution maybe. Each robot have one pose stream， and It just solved my problem of  drawing some signs（primary、text, etc） around it.
```
          "/agv/pose/274030929645667111": {
            "timestamp": 1656000,
            "map_origin": {
              "longitude": 0,
              "latitude": 0,
              "altitude": 0
            },
            "position": [
              6500,
              0,
              3500
            ],
            "orientation": [
              0,
              0,
              0
            ]
          },
          "/agv/pose/274030929645732647": {
            "timestamp": 1660000,
            "map_origin": {
              "longitude": 0,
              "latitude": 0,
              "altitude": 0
            },
            "position": [
              5500,
              0,
              3500
            ],
            "orientation": [
              0,
              0,
              0
            ]
          },
```

```
 {
            "timestamp": 1288000,
            "streams": [
              "/agv/pose/274030929646125863/velocity"
            ],
            "values": {
              "doubles": [
                0
              ]
            }
          },
          {
            "timestamp": 1292000,
            "streams": [
              "/agv/pose/274030929646191399/velocity"
            ],
            "values": {
              "doubles": [
                0
              ]
            }
          }
        ]
      }
```

hookex, 2019-12-06:
I generated data-frame successly. but in `@xviz/parser` model,  only support one  `PRIMARY_POSE_STREAM`, I plan to change it to support `POSE_STREAM_LIST`.

[image]