aurora-opensource/xviz #555: @xviz/server can't install

hookex, 2019-12-05:
```
hooke@DESKTOP-8207B70:/mnt/g/code/hetu.gl$ npm i @xviz/server
npm WARN @megvii/hetu.gl@1.2.0 requires a peer of typescript@>=3.7 but none is installed. You must install peer dependencies yourself.
npm WARN @luma.gl/addons@7.3.2 requires a peer of @luma.gl/core@^7.1.0 but none is installed. You must install peer dependencies yourself.
npm WARN orbit-controls-es6@2.0.1 requires a peer of three@0.108.0 but none is installed. You must install peer dependencies yourself.
npm WARN three.interaction@0.2.2 requires a peer of three@^0.89.0 but none is installed. You must install peer dependencies yourself.
npm WARN @megvii/hetu.gl@1.2.0 No license field.
npm WARN The package @types/node is included as both a dev and production dependency.
npm WARN The package copy-webpack-plugin is included as both a dev and production dependency.
npm WARN The package semantic-release is included as both a dev and production dependency.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm ERR! code ENOENT
npm ERR! syscall chmod
npm ERR! path /mnt/g/code/hetu.gl/node_modules/@xviz/server/bin/xvizserver
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, chmod '/mnt/g/code/hetu.gl/node_modules/@xviz/server/bin/xvizserver'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/hooke/.npm/_logs/2019-12-05T12_19_26_016Z-debug.log

hooke@DESKTOP-8207B70:/mnt/g/code/hetu.gl$ node -v
v12.9.1

hooke@DESKTOP-8207B70:/mnt/g/code/hetu.gl$ npm -v
6.13.1

```

ibgreen, 2019-12-05:
Had this problem quite some time ago. If I recall correctly you need a newer version of the `fsevents` dependency, 1.2.9 is not compatible with node 12. Best case you may be able to just remove your npm lock file and reinstall and it will get automatically bumped.

If not, since this is an indirect dependency it is a slightly trickier to fix. I am not sure about npm, but if you install with `yarn` you can use the `"resolutions"` field in package.json to manually override whatever the installer would normally pick with a newer version that does work with Node 12 and OSX.

ibgreen, 2019-12-05:
Or you could downgrade Node version for now.

twojtasz, 2019-12-05:
The main issue here is the missing scripts directory I believe `/mnt/g/code/hetu.gl/node_modules/@xviz/server/bin/xvizserver`

Let me get on this.

twojtasz, 2019-12-05:
Posted #556

hookex, 2019-12-06:
> Or you could downgrade Node version for now.

Not succes. So I cloned the souce code and use `npm link`.

closed 2019-12-06 by twojtasz

twojtasz, 2019-12-12:
@hookex Can you verify the fix with the latest published @xviz/server@1.0.3

hookex, 2019-12-16:
> @hookex Can you verify the fix with the latest published @xviz/server@1.0.3

Success, good job!

PR #556 (2019-12-06): Fix server module "bin" inclusion with package
Fix #555 and #530