aurora-opensource/xviz #563: Is it possible to send data only once and do not erase in the log view?

didibaba, 2019-12-26:
Is it possible to send data (like point cloud of slam map) only once and do not erase in the log view?

twojtasz, 2020-06-09:
We did add persistence so the specification but the libraries here to not currently expose it.

pai-qi, 2021-05-13:
> We did add persistence so the specification but the libraries here to not currently expose it. I will see when I can work on this and update here.

May I ask if it is done yet?

twojtasz, 2022-01-02:
@pai-qi yes

You can mark a message as `[persistent()](https://github.com/uber/xviz/blob/master/modules/builder/src/builders/xviz-builder.js#L85)`   which has supporting logic in the XVIZStreamBuffer to handle this differently.

I use this in some new scenario examples https://github.com/uber/xviz/tree/master/modules/server/src/scenarios/scenario-arm.js