aurora-opensource/xviz #568: xviz server can not run properly

ytgeng99, 2020-02-07:
After running ./scripts/run-kitti-example.sh, it will throw Error: Cannot find module '../dist/es5'. Node.js version is v12.15.0. Thanks.

KSaikishore, 2020-02-20:
i had the same issue which got resolved by installing jq(Json Parser)

heshiwen1989, 2020-03-10:
> i had the same issue which got resolved by installing jq(Json Parser)

How to Install this, try by npm install jq is error, and try brew install jq and then not find brew; I install brew by npm, and brew not find

twojtasz, 2020-06-09:
@heshiwen1989 What platform are you on?  **jq** is a command line tool that should be available in linux and mac, not sure the setup for windows.

the website shows homebrew support.
https://stedolan.github.io/jq/download/