aurora-opensource/xviz #570: Calling AWS S3 bucket rosbag data

KSaikishore, 2020-02-19:
How do i access rosbag data which is in s3 bucket that doesn't have public access through URL.

ibgreen, 2020-02-19:
This is not a question that XVIZ can help with. 

That said, a typical approach when working with cloud storage such as S3 is to provide an endpoint in your cloud backend that uses AWS apis to create a temporary "signed URL" to your S3 file that you then return to the browser. The browser can then read directly from that signed URL, and you should be able to use that URL with XVIZ. 

The principles are simple though it can be a little fiddly to get everything to work, but since you are clearly building a cloud backend you have already started down that path.

twojtasz, 2020-06-09:
@KSaikishore I think your question as answered in that this is beyond XVIZ and more about how you access your data.  Assuming you have a server and client setup such that they can access each other, the next step is simply using something like the file-reader in the kitti example, which we use in our public demo to read from Github.

 if you have any further questions open up another issue.

closed 2020-06-09 by twojtasz