aurora-opensource/xviz #573: AVS Server Streaming/Live Mode Problem

heshiwen1989, 2020-03-10:
Hello AVS Team: 
      When I  use AVS Streaming/Live by Command: yarn start-streaming-local / yarn start-live-local, the web do not display any content, and capture data by wireshark, There is no any client connect the server.

Orkunkizilirmak, 2020-03-10:
I am having exactly the same issue. When I use "yarn start-streaming" it shows the data that I have in the server but when I try "yarn start-live" page loads up but the data does not visualize.

heshiwen1989, 2020-03-10:
> I am having exactly the same issue. When I use "yarn start-streaming" it shows the data that I have in the server but when I try "yarn start-live" page loads up but the data does not visualize.

Hello, I solve This Problem by Add Parameter in log_from_live.js: 
export default new XVIZLiveLoader({
  logGuid: 'mock',
  bufferLength: 10,
  session_type: 'live',    // +++This is My Add Parameter for live mde ++++
  serverConfig: {
    defaultLogLength: 30,
    serverUrl: 'ws://localhost:8081'
  },
  worker: true,
  maxConcurrency: 4
});

And Then Display is Ok now, And I want To Modified this Server to meet my Application, just like this: 
==========================================================
my data source    ==> | server receiver(New)   AVS server | ===> AVS Client 
==========================================================
or 
===========================================================
my data source(is Server) ====> AVS Client 
===========================================================

Are you Want to do this? If you also want to the same task, we can shared our ideal?

closed 2020-03-10 by heshiwen1989