aurora-opensource/xviz #574: Can offer more detail doc for XVIZ data format?

heshiwen1989, 2020-03-12:
Hello XVIZ Team: 

i plane to use this in realtime, so l need excellent converter to convert my data to xviz, can offer more detail for xviz format?

jlisee, 2020-03-13:
Do the [`Protocol` pages here](https://avs.auto/#/xviz/protocol/protocol-implementations/protobuf-protocol) help?

heshiwen1989, 2020-03-14:
Hello, You offer link: https://avs.auto/#/xviz/protocol/protocol-implementations/protobuf-protocol display is not normal, in other word, this web page do not display any content.

jlisee, 2020-03-15:
That should should be auto-populated from [`docs/protocol-formats/protobuf-protocol.md`](https://github.com/uber/xviz/blob/01fc3b2f2e98e86c904c7b57ee8d0a7be850e770/docs/protocol-formats/protobuf-protocol.md) on GitHub.  I would check the developer console to see what issues you are having with it.

heshiwen1989, 2020-03-18:
Hello, Another Issue: 
 Why your Offer .glb format files of kitti can not correctly parse by tinygltf lib(in c++), and more, why in gltf file, binary data size always greater than buffers' bytelength about 8-11bytes 

e.g.  gltf file parse information: 
magic: "glTF" 
version: 2
length: 2327828
model_length: 95142, in fact this is json data length 
model_format: 0,  but in tiny_gltf, this must be 0x4E4F534A = JSON format.
json data:  "{\"buffers\":[{\"byteLength\":**2232656**}],\"bufferViews\":[{\"buffer\"
binary data: xxxxxx, in total 2232666 bytes

**so that, binary data size > buffers's bytelength**

jlisee, 2020-03-18:
Glad you found the docs, can you open that with a new issue?

closed 2020-03-18 by jlisee