aurora-opensource/xviz #576: AVS Team offer Kitti Demo glb Data Issue

heshiwen1989, 2020-03-18:
Hello AVS Team: 

When parse  your offer .glb format files of kitti is not correctly by tinygltf lib(in c++), and more, I found that the .glb file's binary data size always greater than buffers' bytelength about 8-11bytes

e.g.

gltf file parse information:

magic: "glTF"
version: 2
length: 2327828
model_length: 95142, in fact this is json data length
model_format: 0, but in tiny_gltf, this must be 0x4E4F534A = JSON format.
json data: "{"buffers":[{"byteLength":2232656}],"bufferViews":[{"buffer"
binary data: xxxxxx, in total 2232666 bytes

so that, binary data size > buffers's bytelength

More information:  I analysis the xviz-write module, we find that this module use "encode-glb.js", 
below is "encode-glb.js" javascript code: 

`var MAGIC_glTF = 0x46546c67;
var MAGIC_JSON = 0x4e4f534a;
var MAGIC_BIN = 0x004e4942;
var LE = true;

function encodeGLBSync(glb, dataView) {
  var byteOffset = arguments.length > 2 && arguments[2] !== undefined ? arguments[2] : 0;
  var options = arguments.length > 3 && arguments[3] !== undefined ? arguments[3] : {};
  var _glb$magic = glb.magic,
      magic = _glb$magic === void 0 ? MAGIC_glTF : _glb$magic,
      _glb$version = glb.version,
      version = _glb$version === void 0 ? 2 : _glb$version,
      _glb$json = glb.json,
      json = _glb$json === void 0 ? {} : _glb$json,
      binary = glb.binary;
  var byteOffsetStart = byteOffset;

  if (dataView) {
    dataView.setUint32(byteOffset + 0, magic, LE);
    dataView.setUint32(byteOffset + 4, version, LE);
    dataView.setUint32(byteOffset + 8, 0, LE);
  }

  var byteOffsetFileLength = byteOffset + 8;
  byteOffset += 12;
  var byteOffsetJsonHeader = byteOffset;

  if (dataView) {
    dataView.setUint32(byteOffset + 0, 0, LE);
    dataView.setUint32(byteOffset + 4, MAGIC_JSON, LE);         
  }

  byteOffset += 8;
  var jsonString = JSON.stringify(json);
  byteOffset = (0, _loaderUtils.copyPaddedStringToDataView)(dataView, byteOffset, jsonString, 4);

  if (dataView) {
    var jsonByteLength = byteOffset - byteOffsetJsonHeader - 8;
    dataView.setUint32(byteOffsetJsonHeader + 0, jsonByteLength, LE);
  }

  if (binary) {
    var byteOffsetBinHeader = byteOffset;

    if (dataView) {
      dataView.setUint32(byteOffset + 0, 0, LE);
      dataView.setUint32(byteOffset + 4, MAGIC_BIN, LE);
    }

    byteOffset += 8;
    byteOffset = (0, _loaderUtils.copyPaddedArrayBufferToDataView)(dataView, byteOffset, binary, 4);

    if (dataView) {
      var binByteLength = byteOffset - byteOffsetBinHeader - 8;
      dataView.setUint32(byteOffsetBinHeader + 0, binByteLength, LE);
    }
  }

  if (dataView) {
    var fileByteLength = byteOffset - byteOffsetStart;
    dataView.setUint32(byteOffsetFileLength, fileByteLength, LE);
  }

  return byteOffset;
}`

now, The AVS Team Offer kitti data JSON_MAGIC always is 0x00000000, and then, if we use other tiny_gltf library to parse this file will failed, please check where maybe error, or I have misunderstand this file's format?

ibgreen, 2020-03-18:
> The AVS Team Offer kitti data JSON_MAGIC always is 0x00000000

Are you saying that the code above writes JSON_MAGIC as `0` (there may certainly be a bug but I can't spot it at a glance) or are you referring to some pre-converted test data?

Some additional information:
- The parsing code you pasted is located in loaders.gl https://github.com/uber-web/loaders.gl/blob/master/modules/gltf/src/lib/encode-glb.js 
- if there is indeed a bug in the GLB encoder code the best approach would be to add a failing test case here https://github.com/uber-web/loaders.gl/blob/1.3-release/modules/gltf/test/glb-writer.spec.js  (a PR would be welcome on this)
- As can be seen, right now there are no direct tests for the GLB writer, testing happens indirectly through the GLTFWriter tests: https://github.com/uber-web/loaders.gl/blob/1.3-release/modules/gltf/test/gltf-writer.spec.js

heshiwen1989, 2020-03-18:
> Are you saying that the code above writes JSON_MAGIC as `0` (there may certainly be a bug but I can't spot it at a glance) or are you referring to some pre-converted test data?
>
> Some additional [...]

Hello, 
Yes, I use the demo data which download from github(include kitti and nutonomy?  .glb format data) , and this data have some bytes is error, I gauss this reasons include:  
(1) this data is error
(2) some error when download this data from github  ==> i try to download this data again to check this error

and from this encode-glb.js code, I can not find any error, In fact, I think that the .glb format is like this: 
gltf header   20 bytes  ==> gltf magic, version, file length, json length, json magic   
json data     M bytes  ==> include xviz json data, buffers, bufferviews and so on 
bin header  8 bytes 
bin data      N bytes   ==> pointcloud data and Images data,  and this data is encode or just raw binary data

which is not a normal gltf format data, so i can not use the standard gltf lib to parse this data.

heshiwen1989, 2020-03-19:
Hello, 
I Don't Know How package the glb file, the main difficute is binary data I should how to deal with? it seems that it is not raw data.

heshiwen1989, 2020-03-19:
ok, i solver this issue, thanks!

closed 2020-03-19 by heshiwen1989