aurora-opensource/xviz #580: Error while trying to run ros demo

jediofgever, 2020-04-02:
I met this error when trtying to
` node index.js -d ../../../data/kitti/ --rosConfig `

```
internal/modules/cjs/loader.js:324
      throw err;
      ^

Error: Cannot find module '/home/atas/xviz/node_modules/@xviz/server/dist/es5/index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (internal/modules/cjs/loader.js:316:19)
    at Function.Module._findPath (internal/modules/cjs/loader.js:705:18)
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:969:27)
    at Function.Module._load (internal/modules/cjs/loader.js:864:27)
    at Module.require (internal/modules/cjs/loader.js:1044:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at Object.<anonymous> (/home/atas/xviz/examples/converters/ros/main.js:15:1)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Module._compile (/home/atas/xviz/examples/converters/ros/node_modules/pirates/lib/index.js:99:24)
    at Module._extensions..js (internal/modules/cjs/loader.js:1178:10) {
  code: 'MODULE_NOT_FOUND',
  path: '/home/atas/xviz/node_modules/@xviz/server/package.json',
  requestPath: '@xviz/server'

```

twojtasz, 2020-06-09:
Are you running from your own project with **@xviz/server** as a dependency right.  I'll give it a shot and see if I can reproduce.

Where you on Linux or OSX?

jediofgever, 2020-06-10:
@twojtasz , hi , thanks for reply ,
I solved this issue but actually forget how. 
Now I am interested  in to  visualize object boxes in xviz,
The topic is from ROS with following fields;

```
 uint8 OBJECT_DETECTED=0
uint8 OBJECT_TRACKED=1
uint8 CLASSIFICATION_UNKNOWN=0
uint8 CLASSIFICATION_UNKNOWN_SMALL=1
uint8 CLASSIFICATION_UNKNOWN_MEDIUM=2
uint8 CLASSIFICATION_UNKNOWN_BIG=3
uint8 CLASSIFICATION_PEDESTRIAN=4
uint8 CLASSIFICATION_BIKE=5
uint8 CLASSIFICATION_CAR=6
uint8 CLASSIFICATION_TRUCK=7
uint8 CLASSIFICATION_MOTORCYCLE=8
uint8 CLASSIFICATION_OTHER_VEHICLE=9
uint8 CLASSIFICATION_BARRIER=10
uint8 CLASSIFICATION_SIGN=11
std_msgs/Header header
uint32 id
uint8 detection_level
bool object_classified
geometry_msgs/Pose pose
geometry_msgs/Twist twist
geometry_msgs/Accel accel
geometry_msgs/Polygon polygon
shape_msgs/SolidPrimitive shape
uint8 classification
uint8 classification_certainty
uint32 classification_age
```
 Could you please address any link/steps to achieve this ?