aurora-opensource/xviz #581: Dashed line for polyline

goksanisil23, 2020-04-10:
Hi!
I'm using polyline to create lanes and lanemarkings on streetscape. The lines are appearing as solid lines right now and I was wondering if there is a style property that I could use to make them dashed. 

Thanks and regards

Goksan

twojtasz, 2020-06-09:
@goksanisil23  The current polyline does not support dashed lines. 
 However we do have an example implementation that does.

https://github.com/uber/streetscape.gl/blob/master/modules/layers/src/lane-layer/README.md

This is demonstrated as a custom layer here: https://github.com/uber/streetscape.gl/blob/master/test/apps/layers/src/custom-layers.js

Currently the custom layer approach is a bit awkward because it is out-of-band of the XVIZ message (meaning it's not handled well in the current XVIZ flow to add custom extensions that could be replaced or handled by client code).  So you have to handle it directly.

Since this touches on mapping data which is a bit different from autonomy data in that it is relatively static it's not been a significant area of development. We are getting better in this area  with some of the more recent additions like the scene graph and persistent streams.

closed 2020-06-09 by twojtasz

goksanisil23, 2020-06-09:
Thanks for pointing out the lane layer. It seems that it would do the job for me.