aurora-opensource/xviz #584: How to convert the ROS topics in real time?

zwmisdu, 2020-04-26:
I can successfully use the dataset from kitti to display , and I want to go further. How can I convert the data in ROS topics by XVIZ and use streetscape to display in the form of stream? THANKS.

EliotCao, 2020-04-30:
Maybe xviz/ros docs, ros and ros custom excemple is usefull.

JihunDev, 2020-05-26:
I also want to display the ROS Topic in Real Time. 
I read Doc but couldn't find a way. If you are successful, let me know how.

zwmisdu, 2020-05-26:
> I also want to display the ROS Topic in Real Time.
> I read Doc but couldn't find a way. If you are successful, let me know how.

Sadlly， I haven't achieved this purpose. Maybe I recommend you use ROSNODEJS to pass the information in topics and can deal the information in xx.js.The next thing is that we should convert the message  in XVIZ protocol. And for further, I also need to explore.

filipemtz, 2020-06-23:
I think the python wrapper of xviz could be imported in a ros module and used for publishing data for visualization in streetscape

buni-rock, 2020-06-29:
So far we have managed to create a ros2 subscriber to read data from various topics using rclnodejs. Then the data has been converted to XVIZ protocol but the problem we have encountered now is that we are not able to stream the data to Streetscape. As I have seen, the XVIZ server is able to stream only offline data (data stored in files on hdd). The XVIZ server uses the -d argument for reading the data out of the specified folder. 

How can we stream the data which is already available in NodeJS buffers using XVIZ server?

mrswaffs, 2020-07-23:
you can have a look at #606 that i just created there you have an example scenario in python, however i am currently not able to visualize images with that approach. Other topics work well (you might have to write your own converters).

zwmisdu, 2020-07-24:
> you can have a look at #606 that i just created there you have an example scenario in python, however i am currently not able to visualize images with that approach. Other topics work well (you might have to write your own converters).

Thanks!

JunkyDeveloper, 2020-07-28:
Hello, i currently write a whole xviz server in c++ explizit for live data. No log support will be added because rosbags can replayed and this solve this problem. My current project is with ros1 and the development need a lot of time. Maybe in a few months. My solution will be open source and maybe help you or you can fork this to create add your needed functionality.

aran1177, 2020-09-16:
https://github.com/roarcurry/learn-uber-avs.git