aurora-opensource/xviz #590: issue with Nutonomy converter ,visualisation

KSaikishore, 2020-05-13:
I have converted the nuscenes data(mini,Depreciated datasets) to xviz.But when i visualised that on streetscape.gl camera and lidar data were not getting  visualised properly(data is dissapearing) most of the time.How do i rectify that

twojtasz, 2020-06-09:
Without the ability to reproduce I'm not sure what could be wrong. If you have data you can make accessible and or can record and present some recording or image sequence so I understand what your are experience then I should be able to help.

ElBley, 2020-07-23:
I'm having the same issue, when using the nuscenes conversor with the mini or teaser dataset, when visualizing on streetscape.gl it only properly shows some of the frames. I think this is a conversor issue since most of the .glb files generated are too small (some of the .glb weight around 1.2Mb and the others around 500 kb, probably the heavy ones are the ones that are printing properly).  This clip is from nuscenes v1.0 mini, scene 0061. 
[image]

LeZhang13, 2022-05-11:
I meet the same problem ,it seems only convert the key frame for lidar and image but display all frame for object.

LeZhang13, 2022-05-11:
@twojtasz I have one more question about nuscenes data convert, in nutonomy-converter use the lidar timestamp as the reference timestamp, if i don't want use lidar data and have camera data only, how can I do? Can I use image's timestamp?
```
  // lidar frame is synchronized with ego pose
  // use lidar frame's timestamp as the frame reference timestamp
  _getFrameSensors(frameIndex, sensorsMetadata) {
    const lidar = sensorsMetadata.LIDAR_TOP[frameIndex];
    const timestamp = lidar.timestamp;
    const sensors = {
      LIDAR_TOP: lidar
    };

    const otherSensors = Object.keys(sensorsMetadata).filter(channel => channel !== 'LIDAR_TOP');
    otherSensors.forEach(channel => {
      const samples = sensorsMetadata[channel];
      let i = 0;
      // look for the sample of other sensors with timestamp >= lidar timestamp of current frame
      while (i < samples.length && samples[i].timestamp < timestamp - TIME_WINDOW) {
        i++;
      }
      if (samples[i]) {
        sensors[channel] = samples[i];
      }
    });

    return sensors;
  }
```