aurora-opensource/xviz #593: Python: Working with classes and style_class

sylvaintouret, 2020-05-19:
Hello,

I'm having issue with the python implementation to define style_classes and classes for streams. 

1. Defining classes on a stream, for example in the circle scenario:
```python 
    builder.primitive('/circle').circle([0.0, 0.0, 0.0], self._radius).classes('Bike')
```
To fix this i had to change the primitive.py line 236:
from : 
```python 
    base.classes = self._classes
```
to : 
```python 
    base.classes.append(self._classes)
```
Now this seems to work to apply the classe in the output at least. Now comes the styling for that class. My understanding is that i should be able to do this by building the metadata like this : 

```python
    builder.stream("/circle")\
    .coordinate(xviz.COORDINATE_TYPES.IDENTITY)\
     .stream_style({'fill_color': [200, 0, 70, 128]})\
    .category(xviz.CATEGORY.PRIMITIVE)\
    .type(xviz.PRIMITIVE_TYPES.CIRCLE)\
    .style_class("Car", {'fill_color': [200, 200, 70, 128]})\
    .style_class("Bike", {'fill_color': [0, 200, 70, 128]})
```

I was kinda of able to at least not make it crash by:

1. Adding the build_class_style function in base_builder.py

```python

    def build_class_style(name, style):
        '''
        Create StyleObjectValue from dictionary. It basically deal with list of bytes.
        '''
        data = {
            'name' : name,
            'style' : style
        }
        return StyleClass(**data)
```

2. Replacing the function style_class in metadata.py by this : 

```python
    from xviz_avs.builder.base_builder import build_class_style
    (...)
    def style_class(self, name, style):
        if not self._stream_id:
            self._logger.error('A stream must set before adding a style rule.')
            return self

        stream_rule = build_class_style(name, build_object_style(style))
        self._temp_stream.style_classes.append(stream_rule)
        return self
```

3. Results:

Well, this is where it gets rough for me because my understanding of protobuf still sucks. The result i get is that the class i define for my circle is now visible in the circle tool tip like it should be, but the  circle is now fully white. 

[image]

I hope everything is clear. Any help is welcome on this.

sylvaintouret, 2020-05-19:
It seems to be coming from the color definition itself, because other values seem to be working (like stroke_width)

sylvaintouret, 2020-05-19:
Fixed it by adding this in function _format_primitive in primitive.py. (May hurt sensitive eyes.)

Basically i am manually merging the style from the class to the style from the object. Could not find a better solution for now

```python
    for cls in  self._classes:
        # merge style if class found in streams?
        base.classes.append(cls)
        classes = self._metadata.streams._values[self._stream_id].style_classes._values
        for c in classes:
          if c.name == cls:
                base.style.MergeFrom(c.style)
```

jlisee, 2020-05-21:
@twojtasz

twojtasz, 2020-06-05:
Having worked on the Declarative UI recently, the issues you are running into are due to the use of Protobuf Types directly, and a lack of test coverage.

I'm working through the test coverage issue to expose these cases.

sylvaintouret, 2020-06-30:
Could you include the fix for classes which seems to be working for me 
function _format_primitive in primitive.py:
`[base.classes.append(cls) for cls in  self._classes]`

I've been using it for a while and it does fix at least the class definition even though i have to use stream styling instead of class styling

sylvaintouret, 2020-07-10:
Thanks for the update :)

twojtasz, 2020-07-10:
I'll review this today.

closed 2020-07-10 by twojtasz

kgreenek, 2020-07-10:
Thank you for the quick review cycle!

Would you guys mind doing a minor release to pypi with this bug fix? that way I can just bump the version number instead of having to package a wheel myself to get this in. I'm using it now

twojtasz, 2020-07-10:
I'll try to get a release out today, but not until this evening.

twojtasz, 2020-07-13:
Hitting a test failure i'm fixing first then will release

PR #601 (2020-07-10): Fix python implementation of style_classes
Fixes #593

Currently, attempting to use the classes method results in the following exception when you try to build the metadata message:

```
Assigment not allowed for repeated field "classes" in protocol message
```

Protobuf messages do not allow you to assign to repeated fields directly. You must call extend or append to add items instead.

There was also a protobuf-uesage error in the metadata builder.

Additionally, in order for the style_classes to be converted to json properly, I needed to add logic to unravel them in message.py.