aurora-opensource/xviz #598: Dependency issue when included in streetscape.gl

Algirdyz, 2020-06-17:
Hey,

Wasn't sure where to post this issue, streetscape or xviz. But here goes.

So I've encountered this error today:
https://github.com/visgl/deck.gl/issues/4598

It would seem that I have two versions of @loaders.gl/core in my project when adding @streetscape.gl package.

One of them is @loaders.gl/core@1.3.7 which comes from xviz/parser
Another one is @loaders.gl/core@2.1.6 from @deck.gl/core@8.1.5

I am not really sure how to solve it. Any ideas?

twojtasz, 2020-06-18:
xviz should update to the latest version 2.1.6

Algirdyz, 2020-06-22:
Looking here I see that version range is ^1.3.3
https://github.com/uber/xviz/blob/master/modules/parser/package.json

How do I get it to 2.1.6

twojtasz, 2020-06-22:
I am working to get this tested and push it up today

twojtasz, 2020-06-25:
This is taking a bit longer because the major version bump removed an api we used for handling GLTF data.

twojtasz, 2020-06-26:
I need to extract the code from the old loaders.gl we where using and just embed it in XVIZ, working on it.

In order to *use* the loaders.gl v2, we need to expand the API to allow us to register a custom extension for post-processing of the data.  Considering how Protobuf is more efficient, I don't see a lot of reward for investing in this (gltf support) beyond our current support.

twojtasz, 2020-06-26:
@Algirdyz so just to be clear, your using angular and streetscape along with custom deck.gl layers?  This is the setup for your app?

Algirdyz, 2020-06-28:
Oh no, I am not.
I have a react application with a streetscap component and another component that is based on just deck.gl.
However, from my investigation it just seems that somehow streetscape requires loaders from two different versions. One from xviz/parser and another from deck.gl 
I am not entirely sure how that happened and how it was not a problem before, but it makes sense when I investigate the yarn.lock and that package.json of different.

Algirdyz, 2020-07-21:
Any news regarding this issue?

twojtasz, 2020-07-21:
Nearly there, sorry for the delay.

Should be able to publish tomorrow evening.

Thanks,
Tim

twojtasz, 2020-07-22:
@Algirdyz just published xviz 1.0.6 i'm working to setup a project with streetscape and an additional deck.gl component to verify the fix with the published version

twojtasz, 2020-07-22:
Seems to be working for me.  Let me know if you run into any problems.

closed 2022-04-25 by twojtasz