aurora-opensource/xviz #599: What is the meaning of the message "TODO: Wait protobuf to be fixed" in the xviz/python/xviz_avs/v2/ folder?

filipemtz, 2020-06-23:
Hey there, hope you are fine. I am considering implementing a cpp version of xviz using the python version as an skeleton (btw do you think this is a good idea?). In the README from the folder  xviz/python/xviz_avs/v2/ there is a comment saying "TODO: Wait protobuf to be fixed". Does it mean that the current version of proto files in v2 have problems? What are these problems? The problems affect the python version? What are impacts of the problems? Would it affect the cpp version? 

Hope to hear from you soon.

twojtasz, 2020-06-26:
@filipemtz we have a C++ PR pending already.  See https://github.com/uber/xviz/pull/592

That comment was from the original committer for that Python code so I'm not sure what it refers too.

cmpute, 2020-07-13:
About that comment, I remember it was because a problem of compiling protobuf files at that time, which seems to be fixed by https://github.com/uber/xviz/pull/561. So that TODO can be removed now

cmpute, 2020-07-13:
I didn't check the compilation when answering https://github.com/cmpute/xviz.py/issues/2, but I tried compilation today and it works fine