aurora-opensource/xviz #606: Correct way to convert ros2 images with xviz python?

mrswaffs, 2020-07-23:
Hi there,
I have been trying for a week now to send images from my ros2 application with the python version of xviz to streetscape gl
The part resiving ros2 is done. Then i used the example code and convert the image data to bytes and pass this and the dimensions to the xivizbuilder. However the streetscape server tells me under chrome that it could not decode the image.
With the kitty example and the js version of xviz it works. 
Below is the "anonymised" code for python:

collector code
```python

class Collect_image(Node, ):
   def __init__(self):
        super().__init__('cams')
        self.msg = None
        self.sub_msg = self.create_subscription(Image,  <topic_name>, self.callback, 1)

    def callback(self, msg):
        self.msg = msg

    def get_message(self, builder):
        if self.msg is not None:
            print("get image")
            builder \
                .primitive('/camera/image1') \
                .image(bytes(self.msg.data)) \
                .dimensions(480,270)
```

scenario

```python
import math
import time
import rclpy
import xviz_avs
from xvizServer.scenarios.ros2Collector import Collect_image
import threading
from xviz_avs.builder.declarative_ui import *

class Ros2liveScenario:
    def __init__(self, live=True, radius=30, duration=10):
        self._timestamp = time.time()
        self._radius = radius
        self._duration = duration
        self._live = live
        self._metadata = None
        rclpy.init(args=None)
        self.executor = rclpy.executors.MultiThreadedExecutor(num_threads=4)
        self.image_node = Collect_image()
        t = threading.Thread(target=self.spin_nodes, daemon=True)
        t.start()
        print('finished init')

    def get_metadata(self):
        if not self._metadata:
            builder = xviz_avs.XVIZMetadataBuilder()
            builder.stream("/vehicle_pose").category(xviz_avs.CATEGORY.POSE)
            print("pose done")

            builder.stream('/camera/image_node')\
                .category(xviz_avs.CATEGORY.PRIMITIVE)\
                .type(xviz_avs.PRIMITIVE_TYPES.IMAGE)
            print("image_node done")
            builder.ui(self._get_ui())

            self._metadata = builder.get_message()

        metadata = {
            'type': 'xviz/metadata',
            'data': self._metadata.to_object()
        }

        if not self._live:
            log_start_time = self._timestamp
            metadata['data']['log_info'] = {
                "log_start_time": log_start_time,
                "log_end_time": log_start_time + self._duration
            }

        return metadata

    def spin_nodes(self):
        print("callbacks")
        self.executor.add_node(self.image_node)
        self.executor.spin()

    def get_message(self):
        print("get message")
        builder = xviz_avs.XVIZBuilder(metadata=self._metadata)
        self.image_node.get_message(builder)
        print("msg worked")

        print("all builders done")
        data = builder.get_message()
        print("resived data from builder")
        return {
            'type': 'xviz/state_update',
            'data': data.to_object()
        }

    def _get_ui(self):
        uib = xviz_avs.builder.XVIZUIBuilder()
        panel = XVIZPanelBuilder("image1")
        video = XVIZVideoBuilder(['/camera/image1'])
        panel.child(video)
        uib.child(panel)
```

(For better focus i removed all other topics and messages i convert)
The main and only difference between python and js is that in js a blob with the mimetype (in this case png) is created:
code from the kitty example -- image-converter
``` python  
  xvizBuilder
      .primitive(this.streamName)
      .image(nodeBufferToTypedArray(data), 'png')
      .dimensions(width, height);
  }
```
Therefore I assume the problem is converting the image data to the image primitive correctly i guess.
In my desperation i also tried creating a random number numpy array converted it to uint8 and tried to use this instead of ros images:
```python
        dat = (np.ones(20*80)*127).astype(np.uint8)
        data = Pil_Image.fromarray(dat)
        data = data.convert("RGB")
        im = data.tobytes()
        builder \
            .primitive('/camera/image1') \
            .image(im) \
            .dimensions(20, 80)
```
still no luck 
So if anybody has any ideas please let me know

sidml, 2021-06-22:
@mrswaffs 
Were you able to visualize camera images ?

mr-swaffs, 2021-06-22:
@sidml 
thx for the reminder yes i was able and i have to close the issue.
the posted code actually works with slight adaption: 

I tried to visualize test images i created as numpy matrices, but this is actually rejected by the xviz protocol since they are no
png images -> the header information is missing if you try to directly post the numpy matrix object with the builder.

Instead either directly load images directly from your harddrive using a library of your choice like imageio or pillow
or convert the numpy arrays to pngs using said libraries (Image.fromarray(<your array>))

that does the trick

closed 2021-06-22 by mrswaffs

sidml, 2021-06-23:
@mr-swaffs 
Are you sure that following works ?
`im_bytes = PILImage.fromarray(img.astype(np.uint8)).convert('RGB').tobytes()`

Even loading from a saved file doesn't seem to work!

```python
        im = PILImage.open('test.png')
        w, h = im.width, im.height
        buf = io.BytesIO()
        im.save(buf, format='PNG')
        im_bytes = buf.getvalue()
```
Opencv also doesn't work..

```python
        im = cv2.imread('test.png')
        im_bytes = np.array(cv2.imencode('.png', img)[1]).tobytes()
```
The PNG header (`b'\x89PNG\r\n\x1a\n\x00\x00'`) seems to be present.

I didn't get any error in browser or in python. So, i don't know how to proceed..

sidml, 2021-06-23:
It seems that the following code works
```python
        im = PILImage.open('test.png')
        w, h = im.width, im.height
        buf = io.BytesIO()
        im.save(buf, format='PNG')
        im_bytes = buf.getvalue()
```
I also needed to use python declarative UI. [This comment](https://github.com/uber/xviz/issues/607#issuecomment-790885774) was very helpful