aurora-opensource/xviz #607: How to add time series chart and image display in the circle scenario example in python ?

jrabary, 2020-07-23:
I'm trying to add a time series chart and image (video) visualisation in the python example.
Here are the changes I made so far

in `circle.py` add the stream definition

```
         (builder.stream("/vehicle/velocity")
                 .category(xviz.CATEGORY.TIME_SERIES)
                 .type(xviz.SCALAR_TYPE.FLOAT)
                 .unit('m/s'))
 
             (builder.stream("/vehicle/acceleration")
                 .category(xviz.CATEGORY.TIME_SERIES)
                 .type(xviz.SCALAR_TYPE.FLOAT))
 
             (builder.stream("/camera/front")
                .category(xviz.CATEGORY.PRIMITIVE)
                 .type(xviz.PRIMITIVE_TYPES.IMAGE)
                )
```

and some sample data

```
        builder.time_series('/vehicle/velocity').timestamp(timestamp).value(5.0)
        builder.time_series('/vehicle/acceleration').timestamp(timestamp).value(0.1)
        with open("image.png", "rb") as img_data:
            builder.primitive("/camera/front").image(data=img_data.read())
```

These stream appears in the stream tree of the UI but the chart nor the image as shown below

[image]

How to make them show up in the GUI ?

aalibasic, 2020-08-03:
@jrabary - were you able to resolve this? I am seeing the same issue.

aalibasic, 2020-08-04:
I have some more information as to what is going on.

on my fronend I am using the following code:
https://github.com/uber/streetscape.gl/tree/master/examples/get-started

for the backend, i am using the same code as the author based on this example (plus a single image stream):
https://github.com/uber/xviz/blob/master/python/examples/scenarios/circle.py

When I print the data for my stream for the image on the front end the data is there.

However, I did notice that the python api does not specify the image type (jpg/png) - could this be the source of issues?

[image]

Can anyone provide some guidance on how to add camera data via the python api?

twojtasz, 2020-08-06:
Let me dig into this and get back to you.

aalibasic, 2020-08-10:
@twojtasz - Have you been able to find anything on this front?

Is there information I can provide or something else that I can do the help out with?

twojtasz, 2020-08-10:
Did you add the declarative UI bindings for the chart to those data
streams? That is probably all that is missing.  Let me get an example for
you.

aalibasic, 2020-08-10:
@twojtasz - an example would be awesome - I was not able to find documentation on how to add these bindings.

If I can get your example working, maybe I will add a specific example via a PR on how to get the camera image going so it is more clear to others in the future.

twojtasz, 2020-08-10:
The python Declarative UI bindings is recent.  I have an example but it's
in a different repository.  I'll verify it today and let you know.

twojtasz, 2020-08-10:
@aalibasic @jrabary I have an example Jupyter notebook in another repo

https://github.com/twojtasz/twojtasz.github.io/blob/master/jupyter-notebooks/XVIZ%20Simulated%20Scenario.ipynb

I don't have documentation for this yet, and it expects the latest 0.1.0a4 release for **xviz-avs**.

Let me know if you run into problems or have questions.

aalibasic, 2020-08-17:
@twojtasz - is there a way to do it without using the declarative ui?

We are using the javascript based ui with the python xviz server. Or is there an example you can point me to?

aalibasic, 2020-09-04:
@twojtasz - any update or pointer on this. It would be very appreciated.

TrevorDecker, 2021-01-19:
has anyone gotten this working.  When I attempt it I get an error

 xviz: Unexpected time_series duplicate: /vehicle/velocity
(anonymous) @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
(anonymous) @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
(anonymous) @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
t.parseStreamTimeSeries @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
(anonymous) @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:17
t.default @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:17
y @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
d @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
t.parseXVIZMessageSync @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8
t.onmessage @ 34632506-5d4c-4ddd-9556-23e16f5c6ddf:8

only the first vehicle/velocity is plotted and speed does not change?

hfukada, 2021-03-04:
I know this thread is old and I hate reviving old threads, but: 
```
import xviz_avs.builder
...
# Build your metadata message with the UIBuilder.

            ui_builder = xviz_avs.builder.XVIZUIBuilder()
            graphs = ui_builder.panel('Metrics')
            graphs.child(ui_builder.metric(["/vehicle/velocity"], title="Metrics", description="Vehicle Velocity"))
            ## Add more children for more graphs.
            ui_builder.child(graphs)
            builder = xviz_avs.XVIZMetadataBuilder()   
            builder.stream('/vehicle/velocity').category(xviz_avs.CATEGORY.TIME_SERIES).type(xviz_avs.SCALAR_TYPE.FLOAT).unit('km/h')
```
Is working for me.

The UIBuilder declarative UI is hooking up to the ReactComponent
```
          <XVIZPanel log={log} name="Metrics" />                                                          
```
At least in the "getting started" lab/examples, hooking in your own data requires you to declare the stream_id <-> Metric.name portions as part of the process. 

I also found it confusing how I had to do both react and seemingly python declarative UI, but it's like a container in a container. 
GL

sriselva21-mbrdi, 2022-09-30:
How to add a time series chart and image (video) visualization in the python example, can anyone share sample example code for at least one time series chart with proper documentation/explanation.