aurora-opensource/xviz #609: Error when creating config.json file

Georger2323, 2020-07-24:
I have a .bag file which I want to convert to XVIZ. When I enter the following command, as instructed [by the tutorial ](url):
`./modules/ros/bin/xvizros config path/to/fileName.bag > config.json`

the following error message appears:
```
    throw err;
    ^

Error: Cannot find module '../dist/es5/main'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:580:15)
    at Function.Module._load (internal/modules/cjs/loader.js:506:25)
    at Module.require (internal/modules/cjs/loader.js:636:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/george/xviz/modules/ros/bin/xvizros:51:1)
    at Module._compile (internal/modules/cjs/loader.js:688:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:699:10)
    at Module.load (internal/modules/cjs/loader.js:598:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:537:12)
    at Function.Module._load (internal/modules/cjs/loader.js:529:3)

```

What might be causing this?

twojtasz, 2020-07-24:
That should exist in you built the module.

Can you see if there is a babel-xvizros command in the same folder that
works off the direct source code not transpiled output from the build.

John2397, 2020-07-26:
I'm afraid I didn't understand in which folder you want me to test the `babel-xvizros` command

twojtasz, 2020-07-26:
Modules/Ros/bin

John2397, 2020-07-27:
Yes that file exists in the folder you mentioned.

John2397, 2020-07-27:
After reinstalling a different version of nodejs and npm, it now worked for some reason.

closed 2020-07-27 by John2397