aurora-opensource/xviz #610: Feed data to a XVIZLiveLoader

AdelinaK, 2020-08-04:
Hello. 

I'm working on a project and I want to send data to a XVIZLiveLoader on a web interface. I don't understand exactly how the XVIZServer or XVIZSession are using websockets. I know how to build the data with the XVIZBuilder but I'm confused how the networking part takes place. I wrote the backend in Python. Are there any examples for my case? 

Thank you very much.

susemm, 2020-09-15:
I am also working on this case, unfortunately, don't know how to do.

mhakca, 2020-12-04:
Hi, did you find any solution? @AdelinaK @suseme

AdelinaK, 2020-12-07:
Hi. This example can be useful 
https://github.com/uber/xviz/tree/master/python/examples

You need  XVIZServer, XVIZBaseSession and Builders. XVIZServer contains the websocket connection and the BaseSession runs in a loop and sends data (it is connected to your data source). The Builder is like in the example. 

I hope this helps.

mhakca, 2020-12-18:
Thanks @AdelinaK Did you take data and show it on streetscape side with this method ?