aurora-opensource/xviz #612: How I Can Display PointCloud Map?

heshiwen1989, 2020-08-17:
Hello, 
     I Want Use This Tool Display PointCloud Map, But This Map has Too Many Point(~16M), So I Can Not Send This Map EveryTime: For Example The PointCloud Map Like This: 
[image]

pai-qi, 2021-05-12:
> Hello,
> I Want Use This Tool Display PointCloud Map, But This Map has Too Many Point(~16M), So I Can Not Send This Map EveryTime: For Example The PointCloud Map Like This:
> [image]

Hello, Have you solved this problem yet？

heshiwen1989, 2021-06-16:
> > Hello,
> > I Want Use This Tool Display PointCloud Map, But This Map has Too Many Point(~16M), So I Can Not Send This Map EveryTime: For Example The PointCloud Map Like This:
> > [image]
> 
> Hello, Have you solved this problem yet？

no, because the web side not buffer this map, if you not send everytime, then this map will not display