aurora-opensource/xviz #618: Is custom protobuf.js still required?

ryanjduffy, 2020-09-23:
The git dependency on protobuf.js added by #515 is causing some consternation with our build pipeline. Has the fix been upstreamed and can this dependency be converted to a newer npm version? If not, is there a plan to do so?

octan123, 2021-05-13:
I'm in china, when i install, most time cann't connect private protobuf https://github.com/jlisee/protobuf.js,  #629 also talk this. protobuf.js can repalce by https://github.com/protobufjs/protobuf.js ?

twojtasz, 2022-01-02:
The custom protobuf.js  has tweaks for types arrays.  So it represents a performance improvement not a necessity.  You should be able to use the main version.

octan123, 2022-01-02:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

octan123, 2022-01-05:
@twojtasz thank you very much.