aurora-opensource/xviz #619: Opacity not working

dbruggner, 2020-10-27:
I have a polygon which represents a bounding box for an object. I built this polygon as a flat rectangle and then used extruded:true and height:2.5 as stream style which works fine.
Now I want a transparency effect like in https://avs.auto/demo/ . For this I set the fill_color:"blue", filled:true and opacity:0.4. The bounding box get filled with the color but I don't get the transparency working.
I tested this also in https://avs.auto/playground/, but it doesn't work, too.

I have the information about the opacity style property from here: https://avs.auto/#/xviz/protocol/schema/styling-specification?section=style-properties

Can someone please help me to get this working?

pai-qi, 2021-08-04:
rgba(173, 220, 238, 0.2)，like this

twojtasz, 2022-01-02:
In the playground, I used `    fill_color: '#08f2',` and the box becomes transparent.  

the `opacity' attribute at the stream level does appear to be not working.