aurora-opensource/xviz #625: XVIZFormatWriter is bug?

lamo2k123, 2021-02-04:
Hey! After reading and writing with `xviz / io`, the result is very bad. What am I doing wrong?

source files used https://github.com/uber/xviz-data/tree/master/kitti/2011_09_26_drive_0005_sync

my code:
```
const { resolve } = require('path');

const { XVIZData, XVIZBinaryReader, XVIZFormatWriter, XVIZ_FORMAT } = require('@xviz/io');
const { FileSource, FileSink } = require('@xviz/io/node');

const DIR_DATA = resolve(__dirname, '../_xviz-data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync');
const DIR_TMP = resolve(DIR_DATA, '.tmp');

const fileSource = new FileSource(DIR_DATA);
const fileSink = new FileSink(DIR_TMP);
const birReader = new XVIZBinaryReader(fileSource);
const birWriter = new XVIZFormatWriter(fileSink, { format: XVIZ_FORMAT.BINARY_GLB });

const metadataBuffer = birReader.readMetadata();
const metadataXVIZData = new XVIZData(metadataBuffer);

birWriter.writeMetadata(metadataXVIZData);

const messagesCount = birReader.messageCount();

for(let i = 0; i < messagesCount; i++) {
    const messageBuffer = birReader.readMessage(i);
    const messageXVIZData = new XVIZData(messageBuffer);

    birWriter.writeMessage(i, messageXVIZData);
}

birWriter.close();
```

Origin: https://github.com/lamo2k123/xviz-bug/raw/master/original-file.mov
My result: https://github.com/lamo2k123/xviz-bug/raw/master/read-write-file.mov

twojtasz, 2021-04-20:
Interesting, looks like the color buffer in the points data i getting mutated. I'll try to take a look this week.

twojtasz, 2022-01-02:
I would suggest not using the GLB format (gltf), we are moving away from this as protobuf is often smaller and faster (decoding).

If protobuf shows the same issue please reopen.

closed 2022-01-02 by twojtasz