aurora-opensource/xviz #627: XVIZMetadataBuilder.log_info does not work as intended

yongjun21, 2021-02-24:
According to spec https://avs.auto/#/xviz/api-reference/xviz-builder/xviz-metadata-builder?section=methods,
`XVIZMetadataBuilder.log_info` should allow custom field to be added to `log_info`. However this is constrained by protocol buffer definition as the `LogInfo` message can only have the `start_time` and `end_time` fields

yongjun21, 2021-02-25:
Referring to the Python library

ElenaStrashko, 2021-06-11:
Did you solve this issue?

yongjun21, 2021-06-11:
Using a hack. Add a dummy stream like `/log_info/xyz`. Place the value in the `source` field which accepts free text

ElenaStrashko, 2021-06-11:
Ok, thank you)

twojtasz, 2022-04-25:
Sorry, this only works when using the JSON and the docs are not clear about that. I will keep this open until I update the docs.