aurora-opensource/xviz #634: Python xviz-avs error

shreyas2311, 2021-04-12:
Getting the below error while running serve_scenarios.py

File "serve_scenarios.py", line 8, in <module>
    from xviz_avs import *
  File "/usr/local/lib/python3.8/dist-packages/xviz_avs/__init__.py", line 1, in <module>
    from .builder import XVIZBuilder, XVIZMetadataBuilder,\
  File "/usr/local/lib/python3.8/dist-packages/xviz_avs/builder/__init__.py", line 8, in <module>
    from .base_builder import XVIZBaseBuilder,\
  File "/usr/local/lib/python3.8/dist-packages/xviz_avs/builder/base_builder.py", line 17, in <module>
    (PRIMITIVE_TYPES.CIRCLE, [
AttributeError: 'EnumTypeWrapper' object has no attribute 'CIRCLE'

shreyas2311, 2021-04-14:
even simply importing the package produces error. ie. a file with just the line 
_import xvs_avs_ produces the error.

twojtasz, 2021-04-20:
Odd, i was just using this.  i'll try to verify.  What version of python where you using, and on what platform?

shreyas2311, 2021-04-21:
I am using Ubuntu-20 with python3.

ElenaStrashko, 2021-04-21:
I have the same situation. And I am using Ubuntu 20.1 and Python 3.8.5.

shreyas2311, 2021-04-23:
It seems to work when I run it through anaconda though. However, I got another error in gltf.py where the file was opened in read mode to write into it. Therefore, I downloaded the file from the official [python page](https://pypi.org/project/xviz-avs/) and made the required change which worked.
However, the fix seems to be there in the github repository. This needs to be updated in the official python page as well and the wheel needs to be modified.

closed 2021-05-20 by shreyas2311

sidml, 2021-06-21:
@shreyas2311 
I am getting the same error. How did you resolve it ?

ElenaStrashko, 2021-06-21:
I used Python 3.6.9

sidml, 2021-06-21:
It seems that installing the latest version of protobuf fixes the issue (even in python 3.8.5 and ubuntu 20)
`pip install -U  protobuf`
Then go to the xviz_avs directory and 
`python setup.py install`