aurora-opensource/xviz #639: Adding a new odometer stream for python

shreyas2311, 2021-04-23:
I would like to know how to add a odometer element similar to the acceleration and speed that appears on the circle example. I have tried adding metrics but they only seem render the data in a graphical form.

ekurtgl, 2023-07-12:
Hi @shreyas2311 ,

Have you found a solution to your problem yet?

shreyas2311, 2023-07-12:
No I could not. Eventually we came up with our own UI and made the required  change in the front end.

ekurtgl, 2023-07-12:
I see. I also need a very simple functionality like adding some arbitrary text on top of the objects (similar to object IDs) like this:

[image]

Do you have any idea on how this can be implemented in xviz? Thanks.

shreyas2311, 2023-07-12:
You can use the label primitive if you want to display texts
For example : https://avs.auto/playground/?sample_streams=text

ekurtgl, 2023-07-12:
I was already able to do it on the playground and now I want to do it in my local xviz installation. For that, I modified tracklets-converter.js file under /xviz/examples/converters/kitti/src/converters similar to the playground, but it doesn't seem to have any effect when I run the client. Do you have any guess on what might be the problem? Thank you.

shreyas2311, 2023-07-12:
Oh ok. I am unfamiliar with the JS version. I used the python-xviz-avs. In that, when you are building your UI streams, you would just have to add the same thing and it should work. However, I think it should be something similar for JS as well.

ekurtgl, 2023-07-12:
May I ask which script you use to run the server with python? I followed the readme file under `python` folder, but `serve_scenario.py` throws error:

[image]

Thank you!

shreyas2311, 2023-07-13:
I think I remember having some trouble like this. Did you try running the Circle scenario?

ekurtgl, 2023-07-13:
Oh, apparently it was not a path issue, but I had to install protobuf via `pip install protobuf==3.20.*`. This solved my problem. Thank you!