aurora-opensource/xviz #640: run script ./scripts/run-kitti-example.sh

yanganlan0310, 2021-04-27:
hello!
I'm studying to use xviz and follow the quick start guideline. When I run below script ./scripts/run-kitti-example.sh, I have below error come out. thanks for your help!
environment:
nodejs -- 10.24.1
node --10.24.1
npm --6.14.12
yarn -- 1.22.0
ubutun 18.04 LTS for windows10 (WSL)

(xviz) yang@LAPTOP-GGPCLG7B:~/project/avs/xviz$ ./scripts/run-kitti-example.sh
output:
Generating default KITTI XVIZ data
yarn install v1.22.10
warning package.json: No license field
info No lockfile found.
warning jskitti@1.1.0: No license field
[1/4] Resolving packages...
warning xml2json > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi@13.7.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi > hoek@5.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi > topo@3.0.3: This module has moved and is now available at @hapi/topo. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
warning xml2json > joi > topo > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
warning @babel/cli > @nicolo-ribaudo/chokidar-2 > braces > snapdragon > source-map-resolve > resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
warning @babel/cli > @nicolo-ribaudo/chokidar-2 > braces > snapdragon > source-map-resolve > urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
warning babel-polyfill > core-js@2.6.12: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning babel-polyfill > babel-runtime > core-js@2.6.12: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
[2/4] Fetching packages...
info fsevents@2.3.2: The platform "linux" is incompatible with this module.
info "fsevents@2.3.2" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
[1/3] ⢀ sharp
[2/3] ⢀ node-expat
error /home/yang/project/avs/xviz/examples/converters/kitti/node_modules/node-expat: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments:
Directory: /home/yang/project/avs/xviz/examples/converters/kitti/node_modules/node-expat
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@10.24.1 | linux | x64
gyp info find Python using Python version 3.8.5 found at "/home/yang/anaconda3/bin/python"
gyp info spawn /home/yang/anaconda3/bin/python
gyp info spawn args [ '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/yang/project/avs/xviz/examples/converters/kitti/node_modules/node-expat/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/yang/.cache/node-gyp/10.24.1/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/yang/.cache/node-gyp/10.24.1',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/yang/.cache/node-gyp/10.24.1/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/yang/project/avs/xviz/examples/converters/kitti/node_modules/node-expat',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp ERR! build error
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:13:12)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:19)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:153:21)
gyp ERR! System Linux 4.4.0-19041-Microsoft
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/yang/project/avs/xviz/examples/converters/kitti/node_modules/node-expat
gyp ERR! node -v v10.24.1
gyp ERR! node-gyp -v v5.1.0

yanganlan0310, 2021-04-27:
I meet the same problem! do you solve it?

LeZhang13, 2022-04-26:
you can try to use nvm to change node version to 10.18.0 then retry

twojtasz, 2022-05-02:
The first few issues I see would be 
- it's using microsoft linux, I have not easy way to verify on that environment
- 10.18 is old, I wonder if a newer version works?  I will look into bumping up the version to the oldest supported LTS

twojtasz, 2022-05-02:
The main issue is ```gyp ERR! stack Error: not found: make```

This likely means you are missing the expected build tools for your platform.

I suggest looking at https://askubuntu.com/questions/1037611/development-tools which covers how to install the basic development tools.

prodramp, 2022-05-15:
I can solve the "yarn bootstrap" built errors by using "CXXFLAGS="--std=c++17" yarn bootstrap" command on mac running latest OSX. The problem is that I am keep getting "update required" error when running "./scripts/run-kitti-example.sh" command.

LeZhang13, 2022-05-17:
> 

maybe you can share some capture image