aurora-opensource/xviz #642: Lidar data conversion taking too long

shreyas2311, 2021-05-06:
Hi, I am using the python version of XVIZ AVS. For my requirement, I am converting sensor_msgs/PointCloud2 Lidar data to XVIZ. Presently, I am adding all the points to a list and adding it to the builder. However, when I call builder.get_message, it takes a lot of time. Is there any way to do this faster? I have one minute of data which is taking approximately 75 seconds to convert.

shreyas2311, 2021-05-07:
The MessageToDict function called inside the to_object function of the XVIZFrame class is taking too much time. Therefore, I tried using XVIZProtobufWriter instead of XVIZGLBWriter writer and the file writing is much faster (around 0.0006 sec per frame instead of 0.3 sec). Are there any inconsistencies/issues with this file writer? I am wondering why this was not used as the default way to write files in the circle scenario.

twojtasz, 2021-05-13:
XVIZProtobufWriter is the preferred one really.  XVIZGLBWriter is largely replaced by protobuf.
The performance difference is just that XVIZGLBWriter is not tune, and since it was an early direction for the binary data (really a mix of json & binary) it just doesn't have the same level of perf.

If XVIZProtobufWriter works that should generally give the best performance for generation as well as in-browser perf decoding

shreyas2311, 2021-05-17:
XVIZProtobufWriter faster than XVIZGLBWriter

closed 2021-05-17 by shreyas2311