aurora-opensource/xviz #648: Time-series support for object ID's is not functional

twojtasz, 2021-05-12:
The builder will create them, but when parsing it complains about `Unexpected time_series duplicate`.

It looks like parsing does not consider the object_id at all and reports the duplicate.  And looking at Streetscape.gl it has little to no handling for this either.

shreyas2311, 2021-05-20:
I am trying to have two time series data in the same graph, for example a reference and actual velocity. Is this issue referring to the same as well?

twojtasz, 2021-11-16:
@shreyas2311 The issue i'm referring to is at the data parsing level.  This has been fixed but i'll look at what it would take to have 2 time-series values in the same graph.  I recall we supported this as it is very common to want targer vs measured metrics.

shreyas2311, 2021-11-16:
@twojtasz Yes. I got this to work. This is not an issue. Thanks.

closed 2022-05-02 by twojtasz