aurora-opensource/xviz #663: padTo4Bytes is no longer used in @loaders.gl/loader-utils

xhy279, 2021-08-18:
Hi! 👋 
      
Firstly, thanks for your work on this project! 🙂

Today I used [patch-package](https://github.com/ds300/patch-package) to patch `@xviz/io@1.0.8` for the project I'm working on.

`padTo4Bytes` is not in @loaders.gl/loader-utils anymore. Instead, the refactored function called `padToNBytes`.

Here is the diff that solved my problem:

```diff
diff --git a/node_modules/@xviz/io/dist/esm/gltf/glb-builder.js b/node_modules/@xviz/io/dist/esm/gltf/glb-builder.js
index 1c977e3..f9ddbcd 100644
--- a/node_modules/@xviz/io/dist/esm/gltf/glb-builder.js
+++ b/node_modules/@xviz/io/dist/esm/gltf/glb-builder.js
@@ -1,6 +1,6 @@
 import _classCallCheck from "@babel/runtime/helpers/esm/classCallCheck";
 import _createClass from "@babel/runtime/helpers/esm/createClass";
-import { padTo4Bytes, copyToArray } from '@loaders.gl/loader-utils';
+import { padToNBytes, copyToArray } from '@loaders.gl/loader-utils';
 import { isImage as _isImage } from '@loaders.gl/images';
 import { getAccessorTypeFromSize, getComponentTypeFromArray } from './gltf-utils/gltf-utils';
 import encodeGLBSync from './encode-glb';
@@ -95,7 +95,7 @@ var GLBBuilder = function () {
         byteOffset: this.byteLength,
         byteLength: byteLength
       });
-      this.byteLength += padTo4Bytes(byteLength);
+      this.byteLength += padToNBytes(byteLength, 4);
       this.sourceBuffers.push(buffer);
       return this.json.bufferViews.length - 1;
     }
diff --git a/node_modules/@xviz/io/dist/esm/gltf/parse-glb.js b/node_modules/@xviz/io/dist/esm/gltf/parse-glb.js
index 102d9c1..5bc9e07 100644
--- a/node_modules/@xviz/io/dist/esm/gltf/parse-glb.js
+++ b/node_modules/@xviz/io/dist/esm/gltf/parse-glb.js
@@ -1,5 +1,5 @@
 import '../common/text-encoding';
-import { padTo4Bytes, assert } from '@loaders.gl/loader-utils';
+import { padToNBytes, assert } from '@loaders.gl/loader-utils';
 var MAGIC_glTF = 0x676c5446;
 var GLB_FILE_HEADER_SIZE = 12;
 var GLB_CHUNK_HEADER_SIZE = 8;
@@ -84,7 +84,7 @@ function parseGLBChunksSync(glb, dataView, byteOffset, options) {
       default:
     }
 
-    byteOffset += padTo4Bytes(chunkLength);
+    byteOffset += padToNBytes(chunkLength, 4);
   }
 
   return byteOffset;
```

twojtasz, 2021-11-16:
Interesting, seems like semver was broken. well, i guess it's all alpha.  I"ll get something up this week.  Thanks for reporting.

deerajDev, 2022-02-26:
> ```diff
> padToNBytes
> ```

This resolves the issue.

Thank you so much @xhy279

FFEverglow, 2022-04-08:
Why is my xviz folder empty

YueChuCheng, 2022-04-10:
@xhy279  Hello, sorry I am very new to webpack, I want to ask that after changing padTo4Bytes to padToNBytes, do I need any other command to "update " node_modules, because the warning is still existed after I change the function name.

xhy279, 2022-04-10:
> @xhy279 Hello, sorry I am very new to webpack, I want to ask that after changing padTo4Bytes to padToNBytes, do I need any other command to "update " node_modules, because the warning is still existed after I change the function name.

 It seems like @twojtasz  didn't  fixed it; so u basically either need to fork the @xviz/io repo or use the package I mentioned above patch-package.

YueChuCheng, 2022-04-11:
@xhy279 Thanks for your reply, I currently use patch-package and solve this warning. But now there is another warning while running website-demo, it seems that can't load xviz-data in. Any idea how to fix it? Or how you run website-demo? 
[image]

twojtasz, 2022-04-12:
sorry, i'm spending time this week to address these issue

twojtasz, 2022-04-25:
We have a padTo4Bytes defined in our own code, i've opted to just use that for the time being.

@xviz 1.0.11 published with this fix.

closed 2022-04-25 by twojtasz