aurora-opensource/xviz #667: Has xviz development stalled?

floli, 2021-10-13:
Hey Friends!
I am currently reviewing solutions for visualizations in the automated driving sector and of course, xviz/avs is one of the obvious candidates.

However, I kind of wonder if it's still under development? Last commit is from April and there hasn't been much activity the entire year? Is there anyone know more about the current situation?

Thanks a lot!

ibgreen, 2021-10-13:
Yes. The underlying tech stack (deck.gl, loaders.gl etc) is still actively developed by multiple companies, but xviz and streetscape.gl have lost maintainers and could certainly benefit from some new contributors.

twojtasz, 2021-11-16:
@floli It is incorrect to say this has lost maintainers.  I do actively maintain this project as time and other issues permit.  It is fair to say the public side is largely in maintenance mode currently as other events have limited the amount of active development.

If you ever have any issues or problems I am most active in our slack channel. But as things settle the backlog of issues here will get addressed.

Thanks!

closed 2021-11-16 by twojtasz