aurora-opensource/xviz #673: @xviz/io missing dependencies on @loaders.gl/*

kuta42, 2021-11-17:
`@xviz/io` imports from `@loaders.gl/core`, `@loaders.gl/images`, and `@loaders.gl/loader-utils`, but these packages do not appear as dependencies in its `package.json`. And since the code references `padTo4Bytes` instead of `padToNBytes`, v3 of the  `@loaders.gl` packages can not be used.

ibgreen, 2021-11-17:
Have you tried installing loaders.gl 2.3 instead of 3.0? 

Haven't worked with xviz in a long while, but I would recommend staying on loaders.gl 2.3 until someone has the time to make an official update of xviz to loaders.gl 3.0 (I suspect that there may be other breaking changes beyond the padTo4Bytes issue, we did a lot of changes in 3.0).

twojtasz, 2022-01-02:
I'll try to fix this ASAP.  @ibgreen seems like loaders.gl must have broken semver in the 2.x branch

twojtasz, 2022-01-02:
oh, maybe loaders.gl was alpha at the time this code was written.

twojtasz, 2022-04-25:
Deps where updated, and the issue with padTo4Bytes has been resolved.

closed 2022-04-25 by twojtasz