aurora-opensource/xviz #679: Kitti example is not running

azamshoaib, 2022-01-13:
Hi, thank you for this awesome project. I am running the kitti demo. I have followed the instructions but unfortunately I am getting the error as shown in the figure below

[image]

Kindly help me in this regard. Thank you

SlahserZ, 2022-01-17:
FYI, hope this helps.
https://github.com/uber/streetscape.gl/issues/523

azamshoaib, 2022-01-17:
@SlahserZ Thank you. I will give it a try and let see if it resolves the issue.

lesun90, 2022-02-02:
The problem comes from the new update of loaders.gl package. Follow this fix for the temporary fix.

https://github.com/aurora-opensource/xviz/issues/663

twojtasz, 2022-04-25:
@xviz 1.0.11 fixes this, i'm publish streetscape.gl updates too.

closed 2022-04-25 by twojtasz