aurora-opensource/xviz #684: Convert KITTI to XVIZ - TypeError: (0, _loaderUtils.padTo4Bytes) is not a function

olivia-sp, 2022-01-25:
I'm trying to convert KITTI to XVIZ format
when I run

 yarn start -d ../../../data/kitti/2011_09_26/2011_09_26_drive_0005_sync -o ../../../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync 

then type error occurs,, how can I fix it?

=================================

yarn run v1.22.17
warning package.json: No license field
$ node ./src/index.js -d ../../../data/kitti/2011_09_26/2011_09_26_drive_0005_sync -o ../../../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync
../../../data/kitti/2011_09_26/2011_09_26_drive_0005_sync ../../../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync
Converting KITTI data at ../../../data/kitti/2011_09_26/2011_09_26_drive_0005_sync
Saving to ../../../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync
**D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\dist\es5\gltf\glb-builder.js:111
      this.byteLength += (0, _loaderUtils.padTo4Bytes)(byteLength);
                                                      ^

TypeError: (0 , _loaderUtils.padTo4Bytes) is not a function**
    at GLTFBuilder.addBufferView (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\gltf\glb-builder.js:135:24)
    at GLTFBuilder.addBuffer (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\gltf\glb-builder.js:105:34)
    at packBinaryJsonTypedArray (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:24:35)
    at packBinaryJson (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:58:12)
    at packBinaryJson (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:68:24)
    at map (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:49:36)
    at Array.map (<anonymous>)
    at packBinaryJson (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:49:21)
    at packBinaryJson (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:68:24)
    at packBinaryJson (D:\0_GIT\xviz_kuta\examples\converters\kitti\node_modules\@xviz\io\src\writers\xviz-pack-binary.js:68:24)

Node.js v17.4.0
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

twojtasz, 2022-01-26:
This is a loaders.gl issue, I think a work around is to fix the version.
Let me find the specific details for you

rty813, 2022-04-25:
> This is a loaders.gl issue, I think a work around is to fix the version. Let me find the specific details for you
> […](#)

Is this issue fixed?

twojtasz, 2022-04-25:
Let me fix that in a few hours

twojtasz, 2022-04-25:
@rty813  can you try again after pulling down master, which should be @xviz 1.0.11

twojtasz, 2022-05-02:
This should be fixed. Please re-open if problem persists after pulling down the latest master or using the latest release.

closed 2022-05-02 by twojtasz