aurora-opensource/xviz #708: Style set fill_color in Protobuf Format

jruebsam, 2022-07-22:
I currently try to write some xviz messages in C++ using the protobuf files defined in this repository.
However, I am not getting color settings to work  e.g.  I do something like: 

        auto style = current_poly->mutable_base()->mutable_style();
        std::string fill_color{"#142FF8"};
        style->set_fill_color(fill_color);

but in the streetscape.gl visualizer the color is not properly set.
I assume this is due to the default base64 encoding in the protobuf.js library? 
is there any workaround ? 

Cheers

JunkyDeveloper, 2022-07-22:
I'm also writing a C++ framework for xviz and there it works really good and without any problems. I mostly use it for points.
Which style is that?

jruebsam, 2022-07-25:
This is an object style. Yes this issue only appears when using the protobuf classes, I guess due to some base64 encoding.

jruebsam, 2022-07-28:
Solved by not using color strings instead color arrays

closed 2022-07-28 by jruebsam