aurora-opensource/xviz #709: Extrusion is always into Z direction

jruebsam, 2022-08-09:
When using the XVIZ Polygon class:
If a style has the setting extruded = True,  the extrusion is always happening into the Z direction.
Is this the desired behavior ? I would expect the extrusion to be perpendicular to the defined polygon

jruebsam, 2022-08-16:
I also tried defining every side of the bounding box using a single polygon primitives (so 6 in total ) without the extrusion.
Its Interesting that in this case the sides of the bounding are not drawn at all.
This only happens when the polygons are parallel to the Z-Direction and sounds like a bug to me.

twojtasz, 2022-10-05:
We have used the "define every face" before and it worked.  I'll have to try this in the playground and verify why it might be failing.

The Z direction issue for extrusion is a limitation, as these deck.gl library influenced early primitives and it has a geospatial bias (ie map like ground plane with Z up) and not using the normal of the polygon as the direction.  It probably should use the **normal**, but we would need to update things to get that (ie custom deck.gl layer to use polygon normal not Z-axis).